open-dc-grid/standard #3: 48V bus earthing (grounding)

jlgula, 2020-02-25:
How shall earthing (grounding) be specified on the 48V bus?

Briefly, the issue is that for maximum safety, both poles should be isolated from earth so that someone standing on earth (or in a puddle) won't get electrocuted when touching either pole. This configuration is known as SELV (Separated Extra Low Voltage). Safety authorities generally assert that 30 mA of current flow through a human body is considered an electrocution risk. Under worst case conditions (bare feet on wet ground), 30 mA will flow through typical body resistances when touching 48V. For this reason, DC wiring is limited to a 30V maximum in wet conditions unless it is SELV wired.

This configuration has a problem in that it is difficult to detect an earth fault which would compromise the perceived safety. An undetected single earth fault sets the potential for a double earth fault (both poles earthed) which could potentially start a fire. To minimize this risk, many DC systems deliberately earth one pole, a configuration known as PELV (protected extra low voltage) and require a mechanism to observe current flow through the point of earthing to report the earth fault. Many PV systems are wired this way. 

The following sections reference the attached paper by Chris Moller and are part of an email exchange between Chris Moller and Jim Gula discussing this issue and possible solutions.
[An Earthing Strategy for 48VDC Electrical Ecosystems.pdf](https://github.com/open-dc-grid/standard/files/4251922/An.Earthing.Strategy.for.48VDC.Electrical.Ecosystems.pdf)

Jim:
I have some observations/questions about your grounding paper. I think DC RCD is much more difficult than suggested here. I don’t think the contra-wound transformer works with pure DC. There do exist Type B RCDs that claim to work on pure DC but all the ones that I have seen are very expensive. Some that I have seen inject an AC signal on top the DC and then use the transformer. I did find one ABB paper: [Everything you wanted to know about Type B residual current circuit breakers](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwjj36SCwe3nAhXYop4KHbI1D44QFjAAegQIAxAB&url=https%3A%2F%2Fwww.eevblog.com%2Fforum%2Fbeginners%2Fprobe-oscilloscope-ground-blow-up-issues-usb%2F%3Faction%3Ddlattach%3Battach%3D341471&usg=AOvVaw2_ZcH3W1fGD_1T2dlWcHuV) that briefly describes how DC RCD works using flux saturation and Maxwell’s equation stuff that’s beyond me.

Devices that I’ve seen that are designed to detect earth faults on SELV circuits work on different principals. Typically these inject pulses down the wire and measure what comes back for various distortions. Every manufacturer seems to have a different patented approach.

There’s another problem in that in any long distance connection or in PV panels, there is enough leakage to trip RCD. I ran across another paper, I think it was ABB, that said to not use SELV on runs longer than 100m for this reason.

That’s a reason that I went with the PELV grounding approach with P2030.10. With Open DC Grid (Bruce calls it ODG which I kind of like) we have another chance to revisit these issues. I think we’re adopting the term nanogrid for any wired connection where everyone sees roughly the same voltage. I’m leaning toward making the grounding decision of SELV versus PELV a nanogrid specific decision with the recommendation to use SELV on grids that are entirely inside a structure and PELV for anything that can go outside with isolation recommended (required?) between inside and outside.

Chris:
Many thanks for the feedback.  AC RCDs generate AC for the trip coil, and that’s the problem.  DC can saturate the core, and stop the AC being generated.  A Hall-effect sensor in an air gap instead of a secondary winding will fix the problem if you want to stick with coils (but there’s a possibility of nearby magnetics causing a false trip).
 
If you move away from transformers, I don’t think there’s a need for injecting test signals.  Here’s a very basic design for a DC RCD:
 
[image]

 
If the source is electronic, then it won’t generate fault currents of kiloamps, and sensing resistors can be sensibly small.  (I’ve omitted some detail, like a bridge rectifier on the coil, and components to limit the transient response and set the trip current.)
 
In fact, once you get away from kiloamp fault currents, all kinds of breakers become much easier to design!
 
If leakage to ground is an issue, it will be no different for PELV or SELV.  The difference is that with SELV, you can’t detect it.  Then a second ground fault (eg someone touching a wire) will be dangerous.  PELV allows you to trip on the first fault.
 
The approach I suggested for a single branch circuit safety cutoff is neither PELV nor SELV:
image004.png
[image]

 
Note that the coil shown can actually be an op-amp with as high a resistance to ground as you want.   You may decide that 10mA leakage to ground is OK, but that 30mA could indicate someone getting a dangerous electric shock.
 
To be extra-safe, use both techniques, with the supply grounding trip set to (say) 100mA, and the RCD trip on each branch circuit set to 30mA.

martinjaeger, 2020-02-28:
I really appreciate that you moved the discussion here. And now finally found some time to have a deeper look.

Thanks a lot for the great summary of your findings. I agree that midpoint earthing looks very promising. And also agree that MOSFETs (or solid-state switches in general) should be suitable (and probably most cost-effective) in SELV/PELV systems.

I've got a few open questions for further discussion:

1. How we can implement an earthing system with multiple supplies. The conventional earthing system in AC system is mainly so simple because of the top-down approach of power flow. If we have a DC grid with multiple energy sources (and even worse: in a meshed topology), the section where an error occurred needs to be disconnected by all sources at the same time. Is it sufficient to just implement one of the suggested DC RCDs in each source? Or should we rather have something like an emergency signal that triggers shutdown of all sources if a fault is detected by a centralized RCD?

2. How does the earthing system affect EMC behavior. Can you explain a bit more about the background of your judgement in table 2?

3. And a minor question regarding wire colors as stated in the document: I'm a bit confused about the colors in section 3.0 of Chris' paper. As far as I know, in DC systems the positive wire should be red and the negative either black or blue. Was the blue color for the positive pole a typo? And are there any standards specifying how to choose earth wire colors in DC systems? Could we use the same color (yellow-green) as for AC systems?

Evonet1, 2020-03-04:
Hi Martin, let me take your questions one at a time:
1.	We need to shut down a system (or ideally just a branch circuit) in the event of:
a.	Maintenance work (needs to include lockout capability)
b.	Overcurrent fault
c.	Earth leakage fault
There must also be a well-documented procedure for powering up the system (“black start”).  Startup should be manual, after the maintenance is complete or the fault is cleared.
For a system with multiple power sources (some of which may be intermittent), I think there are only two options to shutting down (and importantly, starting up): 
a)	At higher powers, a safety wire – if this carries usable power (say >10watts), it can be used to drive black-start sequencing electronics – which is extremely desirable – and it can provide diagnostic tools to localise the fault.  Anything on the system can kill everything by momentarily breaking the safety wire or shorting it to ground (but no shutdown sequence in this case).  This is the approach used in the Netherlands (they use 48V).  In my view, some additional work needs to be done on how the safety wire is split between several zones, in the event that you don’t want a fault to black out the whole system.
Renewable energy sources (eg solar panels) must have a way to be told to stop sending power to the system, not just for faults, but also when the battery is fully charged.  AC inverters use rising frequency for this, but this doesn’t work for DC (and voltage is not reliable as an indicator).  A data communications link is probably necessary.
b)	For low-power, low-cost solutions (eg up to 48V or 240watts), an instruction sheet by each safety isolator documenting the sequence of powering down/powering up the power sources, for maintenance.  As long as the battery is closely coupled to the renewable energy source, rising voltage can be used to shut off power inputs when no longer required.  Overcurrent and ground leakage just have to rely on each power source seeing the fault independently.  

2.	Any wire carrying a rapidly changing current will radiate radio waves.  The amplitude is a function of the impedance to ground.  A system that has one wire tied to ground somewhere will radiate less from that wire.  Where both wires are adjacent and have the same impedance to ground, and one has current pulses equal and opposite to the other, the radiated radio waves cancel each other out.  The same is true for external radio waves induced in the wire – the injected noise will be common-mode, and a differential load will not see it.  This is the reason all telephone lines are balanced, as is twisted-pair Ethernet, RS-422 etc.  It’s the essence of what twisted-pair is.  (Parallel wires aren’t quite as good as a twisted pair, but almost.)

3.	No, blue wasn’t a typo.  The proposal is that a wire that carries power and is tied at one point to ground should be coloured blue.   In LVAC, it’s called Neutral (“N”), in LVDC it’s called Midpoint (“M”), but the colour is the same.   In general, as it’s carrying current, there will be a small voltage to ground on it.  A protective equipotential or ground wire should be coloured Green/Yellow, as usual, and must not carry current except in a fault situation.  A wire with a negative DC potential to ground is coloured White, and labelled “–“.  A wire with a positive DC potential to ground is coloured Red, and labelled “+”.  Black isn’t used.  An ungrounded DC system will have Red and White.  If a system is wired ungrounded, and subsequently one wire is grounded somewhere, there is no requirement to change the colour of the grounded wire to blue.

Evonet1, 2020-03-04:
I should have added that a blue positive wire and a white negative wire will only occur in a positive-earth system.  (Incidentally, telephone systems are usually -48VDC, so positive-earth.)