open-dc-grid/standard #6: How to de-energize a bus nanogrid

jlgula, 2020-02-29:
How shall a bus-type nanogrid be de-energized for wiring installation and maintenance?

It seems clear that using semiconductor devices like MOSFETs in place of electro-mechanical devices like DC circuit breakers is important for minimizing costs. Chris Moller posed a question to the IEC LVDC systems committee about whether a system based on MOSFET switches could be used for isolation. They answered by quoting a clause  the generic [IEC safety standard 60364 Electrical Installations for Buildings](https://en.wikipedia.org/wiki/IEC_60364) (the IEC equivalent of the US [NFPA NEC](https://www.nfpa.org/ElectricalSolutions):

*IEC 60364-5-53:2019: 536.2.2.3: Semiconductor devices shall not be used as isolating devices.*

This also relates to a question from Martin earthing systems in grids with multiple supplies. I presume that by “isolation” they mean separation between energized circuits and de-energized circuits.  In strictly a hierarchical grid, the concept of opening a circuit breaker to make wiring safe is relatively straightforward. But in a nanogrid with multiple potential sources, the issue becomes more complicated. I think the use case or context is also important. If normal users are physically isolated from energized circuits (separate issue), then the risk is reduced to people installing or maintaining the wiring. Even in the utility grid, these workers need to use extra caution that normal users don’t have to worry about because of the possibility of back-feeds from PV etc.

I suggest that for bus nano grids we need some kind of keep-alive signal similar what is done with [Sunspec Alliance Rapid Shutdown](https://sunspec.org/rapid-shutdown/). All sources need monitor the bus master and stop supplying power with a short period if they don’t hear from the master. This would permit a single point of shutdown. The master could also monitor one or more RCDs similar to what Chris has suggested. I think we are going to need to do some experiments on Chris’s RCD design to determine if one or more than one is the right solution. I’m concerned about leakage currents generating false triggers. The people at [Voltserver](https://voltserver.com/mobile_communications/)  claim they can detect human touches via disruptions in the pulse flow. It’s possible that if we are using something like IEEE 1901.2 PLC, something similar could be detected but I don’t think I’d want to depend on it.

All of this is using semiconductor devices in significantly more sophisticated ways than just MOSFET isolators. Also, I’m not sure how the IEC 60364 statement reconciles with the existence of [Atom Power solid state circuit breakers](https://www.atompower.com/). An implementation [Karthik Palaniappan’s thesis](https://dc.uwm.edu/etd/2233/) and [another one from Sharthak Munasib](https://scholarworks.uark.edu/etd/1875/). In Googling Atom for recent developments, I see that there is a UL standard titled [UL-489I Outline for Solid State Molded-Case Circuit Breakers](https://www.shopulstandards.com/ProductDetail.aspx?UniqueKey=35950) which Atom claims to satisfy. Wish I had $203 to see what it says. Atom also claims to conform to [UL 1557 - Standard for Electrically Isolated Semiconductor Devices](https://standardscatalog.ul.com/standards/en/standard_1557_6)  available for a mere $402.

One way to reconcile the IEC concern with something that is affordable might be to include a requirement like:

All devices capable of sourcing 48V power shall include a means to physically disconnect the source from the nanogrid. Acceptable means might include:

* DC circuit breaker
* Mechanical switch DC rated for the maximum source current
* Removable fuses on both poles (assuming isolated ground)
* Plug and socket

The idea is to satisfy the IEC requirement inexpensively even though the solution might be awkward and rarely used in practice. For the fuses, I had in mind the automobile ATO-type fuses:

[image]

that could be pulled safely from a live circuit.

martinjaeger, 2020-02-29:
Quick comment regarding the ATO type fuses: Unfortunately they are normally (or always?) only rated for 32V max. There are some versions for 58V, but this seems to be the maximum and not nominal voltage, as they were made for [42V systems](https://en.wikipedia.org/wiki/42-volt_electrical_system) as proposed for cars in the late 90s.

Victron Energy uses [Littelfuse CNN series](https://www.littelfuse.com/products/fuses/specialty-power-fuses/lift-truck-fuses/cnn.aspx) made for forklifts for their DC systems. This might be a viable option, but I didn't investigate how price and availability is compared to ATO blade fuses.

For de-energizing the grid I think it would be better to have a manual switch like in a MCB instead of a fuse that needs to be pulled out, as the fuse might create some sparks that can scare people. The internal actuation of the MCB-type switch doesn't need to be mechanical, but could be with MOSFETs.
A "smart" MCB with a MOSFET could even implement additional functions and not only current interruption. It could for example also trigger during overvoltage events our could even be used as a load switch if it is connected to the communications infrastructure.

jlgula, 2020-02-29:
Good catch on ATO fuses.

I'm looking for a solution that would satisfy the IEC TC64 (LVDC) committee position that "Semiconductor devices shall not be used as isolating devices."

An MCB based MOSFETs would be great but I don't see how to build one that does not conflict with the IEC position. I think some kind of mechanical disconnect is required to satisfy that. DC rated MCBs are expensive. My view is that we need a MOSFET solution as the primary means of disconnect with a mechanical backup to satisfy the IEC requirements.

I did an extensive search for basic on-off switches that would be capable of switching 60V DC at 5A. These are very difficult to find among legitimate suppliers. All most all normal switches are rated for AC or for less than 30V DC. I did eventually find two: a [Bulgin rocker switch (1350) rated at 72V DC 7A $1.52/1K DigiKey](https://www.bulgin.com/products/pub/media/bulgin/data/A_Rocker%201550%201350.pdf) and an [NKK part rated at 72V and 5A](https://www.mouser.com/datasheet/2/295/JW-1663483.pdf).