open-dc-grid/standard #7: Physical isolation on 48V circuits

jlgula, 2020-02-29:
Should the standard require physical isolation between energized circuits and normal users - ie the UL finger test? If so for which nanogrid types?

Most computer equipment is required to conform to UL/IEC 60950  standard that defines various hazards and require separation between hazardous conditions and non-hazardous conditions. It includes the following hazard limits:

* Electric shock - below 60V DC
* Energy - below 240VA

A 48V bus confined to a maximum current of 4A would be considered class III equipment and thus non-hazardous and eligible for exposed circuits like the inside of a PC, (PC with a conforming power supply for AC isolation). We could be more conservative than this and treat the 48V bus as hazardous. This would have implications for interconnect and connectors (if we ever get around to connectors).

Being more conservative might establish good habits for people unused to electricity. We wouldn’t want them to assume that all electricity is as safe as our LVDC circuits and so they should treat all exposed wiring with respect. It’s unclear what imposing this requirement would impose on the cost of standards conforming equipment.

martinjaeger, 2020-02-29:
Does the stated standard limit the current in a 48V grid to 4A in order to be considered non-hazardous? If yes, do you know the reason? 4A would be too low...

jlgula, 2020-02-29:
IEC 60950 section 1.2.8.2 defines various limits on what is considered a DC mains supply. The limits most relevant to us are <60V and less than 240VA. Circuits that fall within those limits are considered “secondary circuits” and subject to less stringent safety requirements. Circuits that exceed those limits are considered hazardous and require restricted access as well as creepage and clearance minimum from secondary circuits. I think that would imply that, for example, a 48V bus with 10A would be considered hazardous. Therefor it would need galvanic isolation from a 12V load circuit that did not have restricted access. If all I/O connections are implemented with restricted access, then there is no problem.

Note a couple of issues. The scope of IEC 60950 is data processing devices which covers PCs, LVDC power supplies and many of the devices we are familiar with. What we are building would probably be considered an energy conversion device which is covered by different standards. Furthermore IEC 60950 is in the process of being replaced by IEC 62368 which defines things differently and includes requirements on energy sources that would cover more of our circuits. I have a copy of 60950 but I don’t have a copy of 62368 so the only information about it that I have is from articles comparing the two standards.

ODG doesn’t necessarily have to conform to these standards but if it didn’t, conforming devices that don’t also meet these standards couldn’t be sold in countries that have strict IEC regulations.

> 
> Does the stated standard limit the current in a 48V grid to 4A in order to be considered non-hazardous? If yes, do you know the reason? 4A would be too low...

James Gula
2018 Galaxy DR
Newport Beach, CA 92660
949-375-2153